UniversalRobots/Universal_Robots_ROS2_Description #35: Add support for UR20

fmauch, 2022-07-11:
As the [UR20](https://www.universal-robots.com/products/ur20-robot/) has been announced, we should add its description. There are still some (internal) things blocking us from adding it, this issue is meant as a single point of reference whenever the question pops up.

pranavVention, 2023-06-26:
Can the URDF be found somewhere?

fmauch, 2023-06-26:
I'm afraid, I'll have to ask for a little more patience...

adamdbrw, 2023-07-28:
@fmauch is there an ETA for this? We would like to simulate UR20 in a nice demo for ROSCon to show in our booth. If we have it, we would show it in a great looking scenario. If we don't have it, we will need to pick another robotic arm.

fmauch, 2023-07-28:
The blocking internal issues seem to be resolved now, so we expect to have this within the next couple of weeks, at least in a usable PR.

adamdbrw, 2023-07-28:
A couple of weeks is alright. If you could let me know in advance in case this is postponed, I would very much appreciate it. We have UR10 currently but it is not fitting for the use-case:
[image]

fmauch, 2023-08-28:
We have been working on this recently and I am positive that we will get something public very soon.

closed 2023-09-05 by fmauch