UniversalRobots/Universal_Robots_ROS2_Description #45: Name of the Links

LucaBross, 2022-11-12:
I try to understand how every link is named.

I made a example picture of my understanding of the names. 
[image]

But I´m not sure abotu the last to links. Is `wirst_3_link` and `flange` the same? 
I already had a Look at the `model.pdf`
[image]
Here the `wirst_1_link` and the `wrist_2_link` is what i just called `wrist_1_link`, right? and the `EE Link` is the `flange`?

When i start the `moveit_config` i saw there is additionaly `base` ,`base_link_interia`, `tool0` and `ft_frame`. 
With `tool0` i can imagine it is a just a empty link in case a tool will be attached. 
`base` and `base_link` is maybe the same?
`base_link_interia` is maybe needed for Gazebo?
But with `ft_frame` i have no clue what it could be.

fmauch, 2022-11-15:
Looking at RViz might be more helpful, in fact

[image]

simply run e.g. `ros2 launch ur_description view_ur.launch.py ur_type:=ur5e` to view the ur5e model. In RViz, you can add a TF display to inspect all the frames. The frames basically follow [this convention](https://gavanderhoorn.github.io/rep/rep-0199.html). The position of a couple of frames are the same, but the orientation might be different.

I'm not be sure TBH, why we still kept the `model.pdf` inside the repo, as it seems to no longer perfectly reflect the URDF.

LucaBross, 2022-11-15:
Ah, okey. Thanks for this. 
In this case the Picture i posted above is right? Or can you find anything thats wrong in this. 
I`m a little bit confused about the shoulderlink and the upper arm link, because their coordinate system start at the the same point.

fmauch, 2022-11-16:
I don't quite remember, but I think that had to do with the COGs and / or the robot calibration.

closed 2022-11-17 by LucaBross