UniversalRobots/Universal_Robots_ROS2_Description #53: Absolute file URIs for Gazebo and Ignition simulations break visualization on remote machine

danzimmerman, 2023-03-17:
I'm trying to work with https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/ to look into [Issue #19](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/issues/19)

### Expected Behavior

After I launch the Gazebo simulation on one computer, I can visualize the robot in RViz on another computer on the same network, similar to the behavior I see with `ur_robot_driver`.

### Actual Behavior

Robot meshes are not visible in RViz on a remote machine because the mesh URIs in `/robot_description` are not `package:///` URIs. Instead, they refer to concrete absolute paths as `file:///` URIs on the simulation machine. 

#### RViz Remote Machine Error Messages
<details>

```
(humble) C:\code\ros>rviz2
[INFO] [1679095199.819414600] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1679095199.819629000] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1679095199.845461900] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1679095199.989519200] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/base.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/base.dae".
[ERROR] [1679095199.989995100] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/base.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/base.dae".
[ERROR] [1679095199.990281000] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/base.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/base.stl
[ERROR] [1679095199.990584900] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/base.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/base.stl
[ERROR] [1679095199.991204100] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/forearm.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/forearm.dae".
[ERROR] [1679095199.991436000] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/forearm.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/forearm.dae".
[ERROR] [1679095199.991629000] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/forearm.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/forearm.stl
[ERROR] [1679095199.991836800] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/forearm.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/forearm.stl
[ERROR] [1679095199.992135200] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/shoulder.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/shoulder.dae".
[ERROR] [1679095199.992327500] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/shoulder.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/shoulder.dae".
[ERROR] [1679095199.992489000] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/shoulder.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/shoulder.stl
[ERROR] [1679095199.992652400] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/shoulder.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/shoulder.stl
[ERROR] [1679095199.992936600] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/upperarm.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/upperarm.dae".
[ERROR] [1679095199.993132700] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/upperarm.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/upperarm.dae".
[ERROR] [1679095199.993289100] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/upperarm.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/upperarm.stl
[ERROR] [1679095199.993527100] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/upperarm.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/upperarm.stl
[ERROR] [1679095199.993817200] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist1.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist1.dae".
[ERROR] [1679095199.994040100] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist1.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist1.dae".
[ERROR] [1679095199.994196200] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/wrist1.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/wrist1.stl
[ERROR] [1679095199.994339600] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/wrist1.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/wrist1.stl
[ERROR] [1679095199.994569800] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist2.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist2.dae".
[ERROR] [1679095199.994753800] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist2.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist2.dae".
[ERROR] [1679095199.994912800] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/wrist2.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/wrist2.stl
[ERROR] [1679095199.995050100] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/wrist2.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/wrist2.stl
[ERROR] [1679095199.995277700] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist3.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist3.dae".
[ERROR] [1679095199.995455600] [rviz2]: Could not load resource [file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist3.dae]: Unable to open file "file:///concrete/path/to/ur_description/meshes/ur5e/visual/wrist3.dae".
[ERROR] [1679095199.995606200] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/wrist3.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/wrist3.stl
[ERROR] [1679095199.995744300] [rviz2]: Error retrieving file [file:///concrete/path/to/ur_description/meshes/ur5e/collision/wrist3.stl]: Couldn't open file /concrete/path/to/ur_description/meshes/ur5e/collision/wrist3.stl
...
```
</details>

`/tf` transforms and other topics are not a problem:

[image]

### Details

This behavior is clearly intentional, per `force_abs_paths` here:

https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/ros2/urdf/ur_macro.xacro#L99

I suppose this is necessary for the URDF -> SDF conversion process to spawn the model in Gazebo Classic or Gazebo (Ignition flavor), like [here](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1272#issuecomment-858407996), but it does break other uses of `/robot_description`.

danzimmerman, 2023-03-18:
I see this is already discussed in https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/pull/15

danzimmerman, 2023-03-18:
I'm using `gazebo_ros_pkgs` version 3.7.0 and it looks like `spawn_entity.py` accepts a `-package_to_model` argument.

In `ur_sim_control.py` I can spawn with

```python
    gazebo_spawn_robot = Node(
        package="gazebo_ros",
        executable="spawn_entity.py",
        name="spawn_ur",
        arguments=["-entity", "ur", "-topic", "robot_description", "-package_to_model"],
        output="screen",
    )
```

I then hardcode `force_abs_paths` to `false` in `ur_macro.xacro` (to make sure the appropriate `ros2_control` content gets loaded) and it seems I get a robot in Gazebo and in RViz on my remote machine.

[image]

danzimmerman, 2023-03-18:
On closer inspection, it doesn't appear like this works properly, or at least not reliably.

danzimmerman, 2023-03-22:
Ah, now I notice that I didn't have `GAZEBO_MODEL_PATH` set correctly. It was pointing to an old colcon workspace.

The `-package_to_model` flag does allow me to visualize in both Gazebo and RViz (local or remote) if I properly add the install location of `ur_description` to `GAZEBO_MODEL_PATH`

Icon45, 2023-05-08:
Hi @danzimmerman

Did you use this package to simulate a Robot with the gazebo-package or is this the default option for the gazebo repo?

Do you know how to use the gazebo-repo with a customized version of this description-repo?

danzimmerman, 2023-05-09:
> Do you know how to use the gazebo-repo with a customized version of this description-repo?

Yes, you could try using the following branches of my forks of this description repo and the Gazebo repo:

Check out the `dz/expose-joint-dyn-prop-file` branch of https://github.com/danzimmerman/Universal_Robots_ROS2_Description

and check out the `dz/expose-joint-dyn-props` branch of https://github.com/danzimmerman/Universal_Robots_ROS2_Gazebo_Simulation

These are works in progress proposing a fix for [#19 in the Gazebo Repo](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/issues/19) by setting the URDF dynamics' tags friction parameter to a nonzero value.

I opened PR #56 for the description. I haven't opened a PR against the Gazebo repo yet.