UniversalRobots/Universal_Robots_ROS2_Description #67: Document usage with moveit - possible issue with joint limits

firesurfer, 2023-05-23:
I have a custom setup with moveit2 where I use the moveit setup assistant to generate the necessary configuration files. I just realized that the generated `joint_limits.yaml` from the setup assistant misses some parts of the `joint_limits.yaml` file provided by the ur description package. 

This makes sense given that the setup assistant creates the `joint_limits.yaml` based on the urdf which can only include certain types of limits. 

When we take for example the `elbow_joint` then the entry in in the limits file provided by the description package looks like this:
(which might also be the most interesting one)
```
  elbow_joint:
    # acceleration limits are not publicly available
    has_acceleration_limits: false
    has_effort_limits: true
    has_position_limits: true
    has_velocity_limits: true
    max_effort: 150.0
    # we artificially limit this joint to half its actual joint position limit
    # to avoid (MoveIt/OMPL) planning problems, as due to the physical
    # construction of the robot, it's impossible to rotate the 'elbow_joint'
    # over more than approx +- 1 pi (the shoulder lift joint gets in the way).
    #
    # This leads to planning problems as the search space will be divided into
    # two sections, with no connections from one to the other.
    #
    # Refer to https://github.com/ros-industrial/universal_robot/issues/265 for
    # more information.
    max_position: !degrees  180.0
    max_velocity: !degrees  180.0
    min_position: !degrees -180.0
```

In the generated file it looks like this (Note I am using a multiarm setup with tf_prefixes set)

```
ur_top/elbow_joint:
    has_velocity_limits: true
    max_velocity: 3.1415926535897931
    has_acceleration_limits: false
    max_acceleration: 0.0
```

fmauch, 2023-05-23:
We had discussions about this in the past, see #62. However, you do have a point that this might have to be documented.

firesurfer, 2023-05-24:
If I understand the changes in #62 correctly then it is a good idea to not use the auto generated `joint_limits.yaml` from moveit.

Additionally I guess I have to add some documentation for how to setup multiarm environments because in those one prob. needs to adapt the limits file by hand.