UniversalRobots/Universal_Robots_ROS2_Description #82: [Question] Intended way of usage

RobinHeitz, 2023-07-24:
Hi, 

I want to integrate a UR5 arm into ros2 humble (previous: Noetic).

I wonder what the intended steps are. I assume:

- Create a custom description package
- copy ur.urdf.xacro into this package and make changes to the file (add end effector, surroundings etc.).
- generate URDF for Setup Assistant with xacro -o my_file.urdf custom_ur.urdf.xacro
- Create Moveit Config Package and use the auto-generated launch files to control the robot with moveit (with use_fake_hardware:=true)

Is that the correct way?
Thanks.

Robin

fmauch, 2023-07-25:
Yes, this seems fairly correct :-)

However, generating a urdf isn't a necessary step as far as I know. To my knowledge, can use the xacro command inside the MSA directly.

fmauch, 2023-08-02:
Just as a comment: I've been successful in generating a moveit config using the MSA on rolling this week. I used the xacro directly with provided parameters and it worked as expected.

RobinHeitz, 2023-08-02:
Thanks for the update.

I'm able to generate those now too. The bug that choosing the save directory is completely gone if starting it inside Docker container ...

RobinHeitz, 2023-08-03:
Well, its not working apparently. I don't know what the issue is, [but I've created a test docker container to reproduce my issue](https://github.com/RobinHeitz/ros2_msa_test).
Clone the repo & build container or just copy the pkg custom_ur_description to reproduce the err.

I've build and source the workspace. Then i launched moveit_setup_assistant -> Create New Package.

In optional xacro arguments I specify the ur_type:
`ur_type:=ur3e` or `ur_type:=ur5e` and select the .xacro file provided in src/custom_ur_description/urdf.

Btw, These xacro files work, can test it with
`ros2 launch custom_ur_description ur_with_environment.launch ur_type:=ur3e`

> The error:
> [INFO] [moveit_setup_assistant-1]: process started with pid [47956]
> [moveit_setup_assistant-1] error: Undefined substitution argument name
> [moveit_setup_assistant-1] when processing file: /home/robin/dev/ros2-msa-test/ws/src/custom_ur_description/urdf/ur_with_environment.xacro
> [moveit_setup_assistant-1] Error:   Error document empty.
> [moveit_setup_assistant-1]          at line 100 in ./urdf_parser/src/model.cpp
> [moveit_setup_assistant-1] Failed to parse robot description using: urdf_xml_parser/URDFXMLParser
> [moveit_setup_assistant-1] [ERROR] [1691048588.764311062] [moveit_setup_assistant.Start Screen]: Error Loading URDF Path. URDF/COLLADA file is not a valid robot model.
> ^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
> [INFO] [moveit_setup_assistant-1]: process has finished cleanly [pid 47956]

I think this error should not occure, because in my launch file i don't specify a name either. But ok, lets pass name:=ur as argument also. I then can create a moveit_config_package (like the one provided in /src).

BUT: If i try to launch any auto-generated launch file, it fails:
`ros2 launch ur3e_test_moveit_config demo.launch.py -d`

> [DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
> [INFO] [launch]: All log files can be found below /home/robin/.ros/log/2023-08-03-09-49-51-505960-rosmachine-48727
> [INFO] [launch]: Default logging verbosity is set to DEBUG
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fbd7857fc10>'
> [DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7fbd7857fc10>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7fbd78ee0100>'
> [DEBUG] [launch]: An exception was raised in an async action/event
> [DEBUG] [launch]: Traceback (most recent call last):
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 161, in _eval_arg
>     return args[name]
> KeyError: 'name'
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 289, in eval_extension
>     return resolve_args(s, context=substitution_args_context)
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 322, in resolve_args
>     resolved = _resolve_args(arg_str, context, commands)
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 338, in _resolve_args
>     resolved = commands[command](resolved, a, args, context)
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 182, in _arg
>     return resolved.replace('$(%s)' % a, _eval_arg(name=args[0], args=context['arg']))
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/substitution_args.py", line 163, in _eval_arg
>     raise ArgException(name)
> xacro.substitution_args.ArgException: name
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 54, in get_launch_description_from_any_launch_file
>     return loader(launch_file_path)
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
>     return getattr(launch_file_module, 'generate_launch_description')()
>   File "/home/robin/dev/ros2-msa-test/ws/install/ur3e_test_moveit_config/share/ur3e_test_moveit_config/launch/demo.launch.py", line 6, in generate_launch_description
>     moveit_config = MoveItConfigsBuilder("ur", package_name="ur3e_test_moveit_config").to_moveit_configs()
>   File "/home/robin/dev/mitsubishi-ros/colcon_melfa/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 510, in to_moveit_configs
>     self.robot_description()
>   File "/home/robin/dev/mitsubishi-ros/colcon_melfa/install/moveit_configs_utils/lib/python3.10/site-packages/moveit_configs_utils/moveit_configs_builder.py", line 232, in robot_description
>     self.__robot_description: load_xacro(
>   File "/opt/ros/humble/lib/python3.10/site-packages/launch_param_builder/utils.py", line 82, in load_xacro
>     file = xacro.process_file(file_path, mappings=mappings)
>   File "/opt/ros/humble/local/lib/python3.10/dist-packages/xacro/__init__.py", line 1098, in process_file
>     process_doc(doc, **kwargs)
[... 22 lines cut ...]
[... cut ...]
> [ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: Undefined substitution argument name
> [DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
> [DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fbd78ee0280>'
> [DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7fbd78ee0280>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7fbd78ee2b90>'

RobertWilbrandt, 2023-08-10:
I am pretty sure this is caused by the starting line:

```
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="$(arg name)">
```

AFAIK the `$(arg name)` is evaluated before the `<xacro:arg [...] />` tags beneath it, causing your substitution errors. Ways to fix this would be either adding an `<xacro:arg name="name" default="ur" />` to your `ur3e_test_moveit_config/config/ur.urdf.xacro` (before the include) or by just statically setting the robot name without a parameter in this place. Both of these work on my machine (leading to new errors about `initial_positions`). I vaguely remember that we also stumbled across this behavior when adjusting the robot names to moveit in the ROS1 driver.

patrickwasp, 2023-09-26:
I was able to load it in moveit2 setup assistant by modifying the `ur.urdf.xacro` file with the following changes:

- `name="$(arg name)"` -> `name="test"`
- `<xacro:arg name="ur_type" default="ur5x"/>` -> `<xacro:arg name="ur_type" default="ur5"/>`

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="test">
   <!-- robot name parameter -->
   <xacro:arg name="name" default="ur"/>
   <!-- import main macro -->
   <xacro:include filename="$(find ur_description)/urdf/ur_macro.xacro"/>

   <!-- possible 'ur_type' values: ur3, ur3e, ur5, ur5e, ur10, ur10e, ur16e, ur20 -->
   <!-- the default value should raise an error in case this was called without defining the type -->
   <xacro:arg name="ur_type" default="ur5"/>
```

It looks like these options are deliberately configured to cause errors for other purposes.

RobinHeitz, 2023-10-11:
OK, it works with this workaround:)
But thats not how it should work, right? Hardcoding the name etc.

RobertWilbrandt, 2023-10-11:
I think this is the correct way to do it. For the general driver description it makes sense to use the robot name as the `<robot>` name, but if you do a custom setup you should probably pick some name that is related to your project. And you are always free to make it configurable yourself - you just have to make sure that the value is passed in through xacro in e.g. the launch file or through MSA.

Seeing that you both were able to generate your setups now, i'll close this issue. Feel free to create a new one if you have further problems.

closed 2023-10-11 by RobertWilbrandt

alesof, 2023-11-01:
Hi there,

During the generation of the MoveIt config package using MSA, I stumbled upon the same error that @RobertWilbrandt  was referring to:

> [...] Both of these work on my machine (leading to new errors about initial_positions).

If you've managed to solve it, it would be great if you could share the solution.