UniversalRobots/Universal_Robots_ROS2_Description #91: Use inertia information from the robot

fmauch, 2023-08-28:
Currently, we use a cylinder approximation for the inertia. However, we can read proper inertia values from the robot. We should probably use this on the description.

mcres, 2024-06-13:
What about simulated robots? Would it be possible to generate more precise inertia values from CAD files?

fmauch, 2024-06-13:
Well yes, this is about robot simulation. The robots contain config files as e.g. the following for a UR5e:

```
[Link]
mass = [3.761, 8.058, 2.846, 1.37, 1.3, 0.365]
center_of_mass = [[0, -0.02561, 0.00193], [0.2125, 0, 0.11336], [0.15, 0.0, 0.0265], [0, -0.0018, 0.01634], [0, 0.0018, 0.01634], [0, 0, -0.001159]]
inertia_matrix = [[0, 0, 0, 0, 0, 0, 0, 0, 0], [0, 0, 0, 0, 0, 0, 0, 0, 0], [0, 0, 0, 0, 0, 0, 0, 0, 0], [0, 0, 0, 0, 0, 0, 0, 0, 0], [0, 0, 0, 0, 0, 0, 0, 0, 0], [0, 0, 0, 0, 0, 0, 0, 0, 0.0002]]

[Geometry]
radius_wrist3       = 0.0375
radius_lower_arm    = 0.0375
offset_x3_lower_arm = 0.0066
```

mcres, 2024-06-13:
Those values are also referenced on a UR Forum post: https://forum.universal-robots.com/t/moment-of-inertia-for-newton-euler-dynamic/14844/2.

I'm a bit confused. Where do those values come from? Is that something stored in the robot controller?

fmauch, 2024-06-19:
> I'm a bit confused. Where do those values come from? Is that something stored in the robot controller?

yes. It is part of `/root/.urcontrol/urcontrol.conf.UR5`. On URSim you will find it in `/ursim/.urcontrol/urcontrol.conf.UR5`

urfeex, 2024-11-21:
This has actually been implemented in https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/pull/195