UniversalRobots/Universal_Robots_ROS2_Description #131: Duplicated xarco "name" argument in the URDFs

macmacal, 2024-01-18:
In both of the provided URDF files (`urdf/ur.urdf.xacro` & `urdf/ur_mocked.urdf.xacro`) there is a requirement for xarco launch argument `name`. However, in the next line this argument is defined by `xacro:arg` marker:

```xml
<robot xmlns:xacro="http://wiki.ros.org/xacro" name="$(arg name)">
  <!-- robot name parameter -->
  <xacro:arg name="name" default="ur"/>
```

As far as I understand the authors' intentions, these URDFs should define and contain all parameters to call the `xacro:ur_robot` macro, without need to pass more information.  Even with provided default value with `xacro:arg` I couldn't parse the file with xacro (on ROS 2 Humble) - the launch argument `name:=` is required. 

I request to clarify this ambiguity.

fmauch, 2024-05-15:
Sorry for the late answer...

That intention is not entirely the case, let me explain:

- The `name` argument is kind of a hack. Since we use one common xacro for all models and we wanted to achieve having the robot name match the model, we need to set the `name` field of the `robot` tag dynamically. It works reading this from a xacro argument, however the default value (when the argument is not provided) is not evaluated at the time when the `robot` tag is parsed.
- The `name` argument is not the only required argument, the `ur_type` argument is also required in a sense that it is used for defining the default values for the model specification. It is set to a default value, however, because it can be omitted if all four robot parametrizations are given.

I hope that explanation helps reasoning the structure as it is. Feel free to comment if you have a suggestion how to handle this differently.

urfeex, 2025-10-01:
I'm going to close this, since the original question seems to have been resolved.

closed 2025-10-01 by urfeex