UniversalRobots/Universal_Robots_ROS2_Description #139: UR10 wrist3 collision and visual mesh are 1mm off

alek5k, 2024-03-08:
There is a 1mm error in both the UR10 collision and visual meshes in the Z direction relative to the origin. This could cause an issue where if you planned a movement in ROS requiring a tolerance of <1mm, a collision could result.

I could not seem to find any corrections for this 1mm in robot kinematics yaml definition.

Evidence (image dimension scaled by 10x):

[image]

Ruoyu-Jia, 2024-04-09:
I met similar problem rencently. The precision of existing dae flies is not good due to reduce the size of files. Is there any way to get more accurate urdf?

fmauch, 2024-04-10:
Thanks for reporting this. The UR10 meshes are rather old, so that probably exists in the ROS1 description, as well.