UniversalRobots/Universal_Robots_ROS2_Description #149: how configure ur5_moveit_config by moveit_setup_assistant?

Asukayoo, 2024-04-11:
I would like to configure my own ur5_moveit_config using moveit_setup_assistant. How can I achieve this? Thank you.

fmauch, 2024-05-08:
Hi there, we are currently working on a comprehensive documentation which is covering that as well. You can look at the preliminary version here: https://docs.universal-robots.com/Universal_Robots_ROS2_Documentation/doc/ur_tutorials/doc/tutorial_index.html

Note that this is valid for rolling and jazzy only. Humble and iron are a bit different, we'll add something for that, as well.

VinDp, 2024-05-29:
Hey @Asukayoo, I will close the issue since some time has passed. Feel free to reopen if still relevant.

closed 2024-05-29 by VinDp