UniversalRobots/Universal_Robots_ROS2_Description #228: Question regarding reuse of license for graphical documentation

firesurfer, 2024-11-15:
Hi, once again I come up with a question about the licensing. 

I was wondering: What is the license of the license for the [graphical documentation](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/rolling/meshes/ur20/LICENSE.txt)

Would it be ok to use the same license text (of course with adapted names etc.) in another project?

As soon as one wants to publish any kind of description packages with meshes one runs into exactly the same problem:
you want to allow the usage of the meshes in the intended way (aka. build your own applications with the robot arm) but you want to ban other usage: e.g. use the mesh to reproduce a certain hardware part yourself - but writing such a license is rather difficult.

gavanderhoorn, 2024-11-15:
IANAL, but I would not be surprised if you couldn't use [Creative Commons](https://creativecommons.org) for this. The [license chooser](https://chooser-beta.creativecommons.org) could provide some guidance, but it's of course not a replacement for a proper lawyer.

CC is not *special* in any way -- it doesn't offer any extra protection against the particular issue you bring up -- but neither is the custom UR license you linked.

If someone is going to ignore the license on your media, I would expect them to do it regardless of whether it's a custom one or CC.

urrsk, 2024-11-20:
@firesurfer please write to legal@universal-robots.com with your request.

closed 2024-11-20 by urrsk