UniversalRobots/Universal_Robots_ROS2_Description #231: Unable to add material[UR20_DIFF_8bit_2K_png_002-material]

JorgePH, 2024-11-29:
Hi,

I am using Jazzy & ur_simulation_gz to simulate an UR20. When launching the default launch with
`ros2 launch ur_simulation_gz ur_sim_control.launch.py ur_type:=ur20`
I get the message `Unable to add material[UR20_DIFF_8bit_2K_png_002-material]` and the UR20 spawns with the blue covers of the upper arm without texture.

[image]

There is a reference in https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/rolling/meshes/ur20/visual/upperarm.dae#L165, but it seems that there is no declaration of the material?

Thanks

urfeex, 2024-11-29:
Thank you for reporting this. The material itself comes from the png file which lies next to that. Possibly GZ Sim doesn't read that one correctly.

urfeex, 2025-01-21:
@JorgePH #244 should contain a fix for the original problem. I would like to finish #243 first, but then that should be ready to merge. If you find the time to test whether that fixes your issue, that's very much appreciated :-)

JorgePH, 2025-01-31:
@urfeex I have tested it in rolling branch from src and I can confirm it solves the problem, the material now renders properly. I have seen that there is already a release including the fix (3.0.0), but the latest binary I can install is the 3.0.0:
```shell
apt list -a ros-jazzy-ur-description
Listing... Done
ros-jazzy-ur-description/noble,now 3.0.0-1noble.20250115.210601 amd64 [installed]
```
Am I missing something?

closed 2025-01-31 by urfeex