UniversalRobots/Universal_Robots_ROS2_Description #257: Visual offset in wrist_2 and wrist_3 for UR3

ShahhhVihaan, 2025-03-04:
I pass on initial_positions for a UR3 arm to `ur.ros2_control.xacro` and I noticed that when the `wrist_3` intial position is 0 `wrist_2` and `wrist_3` seem to be aligned and for other initial positions `wrist_3` looks offset from `wrist_2's` axis. I do not change the urdf from the ur_description package.

Is this an issue with the urdf and if yes, how do we fix it?

[image]
[image]

urfeex, 2025-03-05:
Thank you for reporting this. This is definitively wrong.

ShahhhVihaan, 2025-03-05:
@urfeex May I ask how you generate the xacro files from the CAD model? Do you have internal scripts for cad to urdf generation?

Just curious because the ones that I have seen available online for solidworks, fusion360 or onshape all seem to have certain issues that I always have to work around.

urfeex, 2025-03-05:
This is mainly a manual process, as we don't release new models that often ;-)

In fact, this particular mesh was [added back in 2015](https://github.com/ros-industrial/universal_robot/commits/noetic-devel/ur_description/meshes/ur3/visual/wrist3.dae) and never changed since then (apart from the ambience). So, I assume that this error has been hanging around for quite a while now.

ShahhhVihaan, 2025-03-05:
I see. Thank you.

How can we go about fixing this?

urfeex, 2025-03-05:
@ShahhhVihaan I have just created #258. Could you give that a test?

PR #258 (2025-03-12): Fix UR3 mesh positioning
Fix the positioning of UR3's wrist3 mesh.

Fixes #257
PR #259 (2025-03-12): Fix UR3 mesh positioning (backport #258)
Fix the positioning of UR3's wrist3 mesh.

Fixes #257 <hr>This is an automatic backport of pull request #258 done by [Mergify](https://mergify.com).
closed 2025-03-12 by urfeex