UniversalRobots/Universal_Robots_ROS2_Description #298: Request for UR30 STL or MuJoCo-Compatible Mesh Files

David-Kingsman, 2025-05-27:
I am currently working on  simulating the UR30 robotic arm in MuJoCo. I’ve already obtained the URDF for the UR30, but I noticed that the mesh files (especially STL or MuJoCo-compatible OBJ) seem to be missing or incomplete in the official repository.

Could you kindly provide the STL or OBJ mesh files for the UR30? If a MuJoCo-ready model (including MJCF XML) exists, that would be even more helpful.

Thank you in advance

urfeex, 2025-05-27:
The "missing" mesh files are exactly the same as for the UR20 which is why we didn't duplicate them in the repo. See the [visual_parameters.yaml](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/rolling/config/ur30/visual_parameters.yaml) for details.

urfeex, 2025-09-18:
Closing this as possible solution was posted months ago and no update since then.

closed 2025-09-18 by urfeex